hirotakaster/MQTT #2: Example code fails to compile in Spark IDE

nmelin, 2015-03-10:
when copying the content of the spark library files verbatim ('MQTT.cpp', 'MQTT.h', 'mqtttest.ino'), the compile fails with the error below.  

The same error occurs when forking files directly from https://github.com/hirotakaster/MQTT files.  

Has the spark firmware changed enough to break the example code?
Using SPARK FIRMWARE V0.3.4.   Thanks!

Error Info:
------------------------

"Error: Could not compile. Please review your code."

```
In file included from ../inc/spark_wiring.h:29:0,
from ../inc/spark_wiring_stream.h:36,
from ../inc/spark_wiring_client.h:24,
from ../inc/spark_wiring_tcpclient.h:29,
from MQTT.cpp:56:
../../core-common-lib/SPARK_Firmware_Driver/inc/config.h:12:2: warning: #warning "Defaulting to Release Build" [-Wcpp]
#warning "Defaulting to Release Build"
^
In file included from ../../core-common-lib/CC3000_Host_Driver/evnt_handler.h:38:0,
from ../inc/spark_wlan.h:33,
from ../inc/main.h:38,
from ../inc/spark_utilities.h:30,
from ../inc/spark_wiring.h:33,
from ../inc/spark_wiring_stream.h:36,
from ../inc/spark_wiring_client.h:24,
from ../inc/spark_wiring_tcpclient.h:29,
from MQTT.cpp:56:
../../core-common-lib/CC3000_Host_Driver/socket.h:146:0: warning: "fd_set" redefined [enabled by default]
#define fd_set _types_fd_set_cc3000
^
In file included from /opt/gcc_arm/arm-none-eabi/include/stdio.h:47:0,
from ../inc/spark_wiring_print.h:30,
from ../inc/spark_wiring_string.h:33,
from MQTT.cpp:55:
/opt/gcc_arm/arm-none-eabi/include/sys/types.h:256:0: note: this is the location of the previous definition
#define fd_set _types_fd_set
^
In file included from ../../core-common-lib/CC3000_Host_Driver/evnt_handler.h:38:0,
from ../inc/spark_wlan.h:33,
from ../inc/main.h:38,
from ../inc/spark_utilities.h:30,
from ../inc/spark_wiring.h:33,
from ../inc/spark_wiring_stream.h:36,
from ../inc/spark_wiring_client.h:24,
from ../inc/spark_wiring_tcpclient.h:29,
from MQTT.cpp:56:
../../core-common-lib/CC3000_Host_Driver/socket.h:162:0: warning: "FD_SET" redefined [enabled by default]
#define FD_SET(fd, fdsetp) __FD_SET (fd, fdsetp)
^
In file included from /opt/gcc_arm/arm-none-eabi/include/stdio.h:47:0,
from ../inc/spark_wiring_print.h:30,
from ../inc/spark_wiring_string.h:33,
from MQTT.cpp:55:
/opt/gcc_arm/arm-none-eabi/include/sys/types.h:258:0: note: this is the location of the previous definition
# define FD_SET(n, p) ((p)->fds_bits[(n)/NFDBITS] |= (1L << ((n) % NFDBITS)))
^
In file included from ../../core-common-lib/CC3000_Host_Driver/evnt_handler.h:38:0,
from ../inc/spark_wlan.h:33,
from ../inc/main.h:38,
from ../inc/spark_utilities.h:30,
from ../inc/spark_wiring.h:33,
from ../inc/spark_wiring_stream.h:36,
from ../inc/spark_wiring_client.h:24,
from ../inc/spark_wiring_tcpclient.h:29,
from MQTT.cpp:56:
../../core-common-lib/CC3000_Host_Driver/socket.h:163:0: warning: "FD_CLR" redefined [enabled by default]
#define FD_CLR(fd, fdsetp) __FD_CLR (fd, fdsetp)
^
In file included from /opt/gcc_arm/arm-none-eabi/include/stdio.h:47:0,
from ../inc/spark_wiring_print.h:30,
from ../inc/spark_wiring_string.h:33,
from MQTT.cpp:55:
/opt/gcc_arm/arm-none-eabi/include/sys/types.h:259:0: note: this is the location of the previous definition
# define FD_CLR(n, p) ((p)->fds_bits[(n)/NFDBITS] &= ~(1L << ((n) % NFDBITS)))
^
In file included from ../../core-common-lib/CC3000_Host_Driver/evnt_handler.h:38:0,
from ../inc/spark_wlan.h:33,
from ../inc/main.h:38,
from ../inc/spark_utilities.h:30,
from ../inc/spark_wiring.h:33,
from ../inc/spark_wiring_stream.h:36,
from ../inc/spark_wiring_client.h:24,
from ../inc/spark_wiring_tcpclient.h:29,
from MQTT.cpp:56:
../../core-common-lib/CC3000_Host_Driver/socket.h:164:0: warning: "FD_ISSET" redefined [enabled by default]
#define FD_ISSET(fd, fdsetp) __FD_ISSET (fd, fdsetp)
^
In file included from /opt/gcc_arm/arm-none-eabi/include/stdio.h:47:0,
from ../inc/spark_wiring_print.h:30,
from ../inc/spark_wiring_string.h:33,
from MQTT.cpp:55:
/opt/gcc_arm/arm-none-eabi/include/sys/types.h:260:0: note: this is the location of the previous definition
# define FD_ISSET(n, p) ((p)->fds_bits[(n)/NFDBITS] & (1L << ((n) % NFDBITS)))
^
In file included from ../../core-common-lib/CC3000_Host_Driver/evnt_handler.h:38:0,
from ../inc/spark_wlan.h:33,
from ../inc/main.h:38,
from ../inc/spark_utilities.h:30,
from ../inc/spark_wiring.h:33,
from ../inc/spark_wiring_stream.h:36,
from ../inc/spark_wiring_client.h:24,
from ../inc/spark_wiring_tcpclient.h:29,
from MQTT.cpp:56:
../../core-common-lib/CC3000_Host_Driver/socket.h:165:0: warning: "FD_ZERO" redefined [enabled by default]
#define FD_ZERO(fdsetp) __FD_ZERO (fdsetp)
^
In file included from /opt/gcc_arm/arm-none-eabi/include/stdio.h:47:0,
from ../inc/spark_wiring_print.h:30,
from ../inc/spark_wiring_string.h:33,
from MQTT.cpp:55:
/opt/gcc_arm/arm-none-eabi/include/sys/types.h:261:0: note: this is the location of the previous definition
# define FD_ZERO(p) (__extension__ (void)({ \
^
In file included from ../inc/spark_wiring.h:29:0,
from ../inc/application.h:29,
from mqtttest.cpp:2:
../../core-common-lib/SPARK_Firmware_Driver/inc/config.h:12:2: warning: #warning "Defaulting to Release Build" [-Wcpp]
#warning "Defaulting to Release Build"
^
mqtttest.cpp:5:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
#line 1 
^
mqtttest.cpp: In function 'void callback(char*, byte*, unsigned int)':
mqtttest.cpp:11:15: warning: converting to non-pointer type 'char' from NULL [-Wconversion-null]

^
mqtttest.cpp: In function 'void setup()':
mqtttest.cpp:30:33: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

^
mqtttest.cpp:34:48: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
// connect to the server
^
mqtttest.cpp:34:48: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
mqtttest.cpp:35:35: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
client.connect("sparkclient");
^
mqtttest.o: In function `setup':
/spark/compile_server/shared/workspace/2_compile-server2/core-firmware/build/mqtttest.cpp:30: undefined reference to `MQTT::connect(char*)'
/spark/compile_server/shared/workspace/2_compile-server2/core-firmware/build/mqtttest.cpp:33: undefined reference to `MQTT::isConnected()'
/spark/compile_server/shared/workspace/2_compile-server2/core-firmware/build/mqtttest.cpp:34: undefined reference to `MQTT::publish(char*, char*)'
/spark/compile_server/shared/workspace/2_compile-server2/core-firmware/build/mqtttest.cpp:35: undefined reference to `MQTT::subscribe(char*)'
mqtttest.o: In function `loop':
/spark/compile_server/shared/workspace/2_compile-server2/core-firmware/build/mqtttest.cpp:40: undefined reference to `MQTT::isConnected()'
/spark/compile_server/shared/workspace/2_compile-server2/core-firmware/build/mqtttest.cpp:41: undefined reference to `MQTT::loop()'
mqtttest.o: In function `__static_initialization_and_destruction_0':
/spark/compile_server/shared/workspace/2_compile-server2/core-firmware/build/mqtttest.cpp:5: undefined reference to `MQTT::MQTT(char*, unsigned short, void (*)(char*, unsigned char*, unsigned int))'
collect2: error: ld returned 1 exit status
make: *** [b43d8f930b22f0b9b2e8154ddd3da59ac91e0edc109424a182be4a17143d.elf] Error 1

```

hirotakaster, 2015-03-11:
Hi nmelin,
I update spark.json file Library version to 0.2.0 now, will publish to Spark.io IDE.
This problem was already merged, check the following.
https://github.com/hirotakaster/MQTT/commit/922065b2b3d1ae35c37fa2bc37370f35c31a03b0
Thanks.

nmelin, 2015-03-11:
Hello @hirotakaster , 

I see the `#include` has changed from `/MQTT.h` to `MQTT/MQTT.h`.
Is that the only change?

I included this change in the `MQTTTEST.INO` file, but the code still fails to compile in the Spark IDE.

Also the MQTT library inside the spark IDE is still listed as "0.1.0".

Do you have any ideas about why the code fails to compile in the spark IDE?

I'm hoping to assist with testing, but I don't understand C well enough to understand why the compile is failing.

Thanks.

hirotakaster, 2015-03-11:
Hi nmelin,
I don't get compile error on Spark IDE Web now. check the next on Spark IDE web.
1.  Libraries -> MQTT
2. click MQTTTEST.INO and "USE THIS EXAMPLE"
3. #include "MQTT.h" to #include "MQTT/MQTT.h"
4. click Verify button.
may be you can compile with no error.
I don't know why version can't update, I will ask spark community.
Thanks.

hirotakaster, 2015-03-27:
Hi @nmelin, 
I finally update MQTT library to 0.2.0 on Spark WebIDE. You can easy to use sample source code.
Thanks.

hirotakaster, 2015-03-27:
Hi @nmelin, 
I finally update MQTT library to 0.2.0 on Spark WebIDE. You can easy to use sample source code. Please check and close this issue.
Thanks.

closed 2016-01-13 by hirotakaster