hirotakaster/MQTT #3: spark can't connect to cloudmqtt

milonet, 2015-04-22:
Hi, i have many problem to connect the spark to cloudmqtt. I'm using your example (mqtttest.ino)
suppose that my complete cloud mqtt url is: "mqtt://username:password@host:port"
where i can put the connection string in your code?

here ? if yes, how?
MQTT client("server_name", 1883, callback);

or, (from instructables.com) here? if yes, how?
-----
In the "mqtttest.ino" Line 29 : 
client.connect("sparkclient");
change it to :
client.connect("cloudmqtt-url", "userid", "password");
-----

thank you

hirotakaster, 2015-04-22:
before connect, set server ip(or domain) like this.
  MQTT client("mxx.cloudmqtt.com", 13301, callback);
and next, connect to the server.
  client.connect("sparkclient", "id", "password");
You can set the ID(sparkclient) that you like. Now I check the Spark MQTT with cloudmqtt, there is no problem to use.

milonet, 2015-04-23:
ok, works!! thank you!

closed 2015-04-23 by milonet