hirotakaster/MQTT #5: initializing argument 2 of 'bool MQTT::publish(char*, char*)' [-fpermissive]

dande011, 2015-07-02:
Veryfing code using this library in Particle Build fails with this error:
```
MQTT/MQTT.h:113:9: error:   initializing argument 2 of 'bool MQTT::publish(char*, char*)' [-fpermissive]
    bool publish(char *, char *)

```
Here is the line that is causing it trouble:
```
client.publish("hello/world", (topic == "/home/test")?"TRUE":"FALSE");
```
I don't know if the conditional operator is doing something unexpected, but it should just show up as "TRUE" or "FALSE", as far as the function call can see.

hirotakaster, 2015-07-02:
let's try like this. 

  client.publish("hello/world", (topic == "/home/test") ? (char *)"TRUE" : (char *)"FALSE");

jeffeb3, 2015-11-15:
These arguments should be "const char *", not "char *" There's no reason you would need to edit them. This should be changed in the library.

hirotakaster, 2015-11-17:
now update to 7e711af0d0d332dc23e5b1d681a0fe00a4500b90

closed 2016-01-13 by hirotakaster