hirotakaster/MQTT #11: MQTT/MQTT.h No Such File

mtngit14, 2015-10-30:
MQTT test firmware compiles (and flashed Photon) in Particle cloud but not in desktop Particle Dev.
Changing the include to just #include "MQTT.h" allowed compiling/flashing in Particle Dev. 
Note program and includes all located in same flat directory.
Could someone validate this change as a correct solution and not really changing some other parameter.
Particle Dev on MacBook Pro with El Capitan 10.11.1 

David
RavenTT / MiqroMeq

timpulver, 2015-11-02:
Had the same issue with latest Particle Dev on Mac OS X El 10.11.1 and the Neopixel library – putting all files into the same folder as the main file (no subfolders) worked.

hirotakaster, 2015-11-04:
I think this problem is based Spark Deve desktop client.

closed 2016-01-13 by hirotakaster