hirotakaster/MQTT #12: subscribe() fails when giving MQTTQOS1 as second argument

Lahorde, 2015-11-02:
When calling :

    mqttClient.subscribe(topic, MQTTQOS1);

no messages are received on topic, changing to MQTTQOS0 works fine.

hirotakaster, 2015-11-04:
Hi Lahorde,
I recommend you implement re-connect method or connect to the mqtt-server when you need. In my case, when keeping connect mqtt client with server suddenly disconnect or tcp connection is fail. That's reason is not based mqtt client library, I think photon firmware bug or wifi connection problem.
And FYI, as you know mqtt use tcp connection on wifi, keeping the tcp connection is high power consumption in my test on photon. so I recommend you would be better connect to mqtt server when you need only.

Lahorde, 2015-11-06:
Hi hirotakaster, 
As you can see I already reconnect to mqtt when connection drops (loop() code :
https://github.com/OpHaCo/roomba_wifi/blob/master/roomba-control.ino
I lost some packets because of frequents connection drops. But before discovering it I tried to modify qos and saw this issue. No I have no more disconnections and performed some tests checking no packets are lost. 

Both :

hirotakaster, 2015-11-17:
oh, I see. please send a pull request and contribute your commit.
thank you.

closed 2016-01-13 by hirotakaster