hirotakaster/MQTT #13: MQTT client disconnects when connecting mosquitto 1.4.3

Lahorde, 2015-11-03:
When using MQTT library, I lost some notifications. I added logs and discovered MQTT client often disconnects. I have another client that is connected to broker on same topic : it does not disconnect.

Here is code that runs on photon : https://github.com/OpHaCo/roomba_wifi/blob/master/roomba-control.ino
Mosquitto server is :

    mosquitto version 1.4.3 (build date 2015-10-02 12:31:25+0200)

Here are mosquitto logs :

```
nov. 03 10:59:50 remi-hp9470m mosquitto[548]: 1446544790: Client roomba_photon disconnected.
nov. 03 10:59:50 remi-hp9470m mosquitto[548]: 1446544790: New client connected from 192.168.132.63 as roomba_photon (c1, k15).
nov. 03 10:59:50 remi-hp9470m mosquitto[548]: 1446544790: New connection from 192.168.132.64 on port 1883.
nov. 03 10:59:50 remi-hp9470m mosquitto[548]: 1446544790: Client roomba_photon already connected, closing old connection.
nov. 03 10:59:50 remi-hp9470m mosquitto[548]: 1446544790: Client roomba_photon disconnected.
nov. 03 10:59:50 remi-hp9470m mosquitto[548]: 1446544790: New client connected from 192.168.132.64 as roomba_photon (c1, k15).
nov. 03 10:59:50 remi-hp9470m mosquitto[548]: 1446544790: New connection from 192.168.132.63 on port 1883.
nov. 03 10:59:50 remi-hp9470m mosquitto[548]: 1446544790: Client roomba_photon already connected, closing old connection.
nov. 03 10:59:50 remi-hp9470m mosquitto[548]: 1446544790: Client roomba_photon disconnected.
nov. 03 10:59:50 remi-hp9470m mosquitto[548]: 1446544790: New client connected from 192.168.132.63 as roomba_photon (c1, k15)
```
Here are photon logs :

```
connected to mqtt
mqtt client not connected - id=42938
connect mqtt
connected to mqtt
mqtt client not connected - id=43127
connect mqtt
connected to mqtt
mqtt client not connected - id=43314
connect mqtt
connected to mqtt
mqtt client not connected - id=43499
connect mqtt
connected to mqtt
mqtt client not connected - id=43680
connect mqtt
connected to mqtt
mqtt client not connected - id=43864
connect mqtt
connected to mqtt
mqtt client not connected - id=44056
connect mqtt
connected to mqtt
mqtt client not connected - id=44241
connect mqtt
connected to mqtt
mqtt client not connected - id=44430
connect mqtt
connected to mqtt
mqtt client not connected - id=44623
connect mqtt
connected to mqtt
mqtt client not connected - id=44812
connect mqtt
connected to mqtt
mqtt client not connected - id=44999
connect mqtt
connected to mqtt
mqtt client not connected - id=45195
connect mqtt
connected to mqtt
mqtt client not connected - id=45380
connect mqtt
connected to mqtt
mqtt client not connected - id=45572
connect mqtt
connected to mqtt
mqtt client not connected - id=45760
connect mqtt
connected to mqtt
mqtt client not connected - id=45940
```

I tried with iot.eclipse.org and 

    mosquitto version 1.2.3 (build date 2015-10-07 16:19:36+0200)
and it works fine. No messages are lost, photon client is not disconnected.

hirotakaster, 2015-11-04:
Hi, Lahorde.
I tyr to use mosquitto 1.4.4(latest, mosquitto version 1.4.4 (build date 2015-11-04 19:18:23+0900)) now, photon with mqtt mqtttest.ino(check mqtt example on photon) works well.
may be you use same use id on pub/sub? If use same use id, can't connect to mosquitto.

Lahorde, 2015-11-06:
Hi hirotakaster,

I've just performed some new tests and I do not have anymore any disconnections with MQTT 1.4.3. I did not use same id on pub/sub. There is just 1 difference in setup, I moved to another wifi network mosquitto server and photn.

closed 2015-11-06 by Lahorde