hirotakaster/MQTT #17: TLS

jpmens, 2015-12-23:
Thank you very much, @hirotakaster, for this library; I've been using this now for a few weeks, and it looks very solid.

Is there any chance we could convince you to add (optional) TLS support to it? I  believe that, in this day and age, we should communicate securely whenever possible.

hirotakaster, 2015-12-27:
Hi jpmes,
Oh, your contribution are very helpful and welcome!! SSL/TLS security option is needed in MQTT, but Photon/Spark firmware don't have SSL/TLS library yet. So I think If I implement SSL/TLS with MQTT,  simple SSL/TSL(socket library like a OpenSSL) have to include to the MQTT(Photon) source code tree.

jpmens, 2015-12-27:
Indeed, but I think OpenSSL would be much too large. I recently stumbled over wolfSSL (https://www.wolfssl.com/wolfSSL/Home.html) which seems much more suited to embedded hardware.

hirotakaster, 2015-12-27:
Thank you. I know wolfSSL now, this is good for embedded system like a Photon.

baijerry, 2016-07-18:
Hi @hirotakaster , any updates on TLS support? Over in the community forums there is buzz about glowfish's httpsclient (https://github.com/glowfishAPI/httpsclient-particle#adhere-to-security) as well as https://tls.mbed.org/

hirotakaster, 2016-07-19:
Hi, is this topic? https://community.particle.io/t/https-client-is-here-for-the-photon-by-the-glowfi-sh-team/15934
httpsclient-particle license is based GPL so, maybe hard to use. If you try implement TLS library with MQTT, you would be better to use TropicSSL or PolarSSL.

mpchst, 2016-11-09:
mbed TLS (previously known as PolarSSL) is used in the Photon firmware already: https://github.com/spark/firmware/tree/develop/communication/lib/mbedtls

And using it for MQTT should be feasable as well: https://github.com/MediaTek-Labs/aws_mbedtls_mqtt

hirotakaster, 2016-11-09:
Hi @mpchst , thank you for your info. I known mbedTLS  in firmware, but Particle's mbedtls header files can't use directly, it need to include all mbedtls files(header/source) into library. MediaTeck-Labs implements are very useful reference.

hirotakaster, 2016-11-11:
I try to use Particle mbedTLS, but it's hard to use and having some problems. I use Particle mbedTLS on WebIDE/Desktop IDE, but it can't compile. This issue need long time:)

hirotakaster, 2016-11-19:
Hi @mpchst , @jpmens , Now I publish TlsTcpClient : https://github.com/hirotakaster/TlsTcpClient , and I will implements this library to MQTT.

hirotakaster, 2016-11-28:
Hi @jpmens , @baijerry , @mpchst , Now I upload MQTT-TLS library 1st version.
https://github.com/hirotakaster/MQTT-TLS
TLS library is very fat, so I upload to other repo.

mpchst, 2016-11-30:
Great news, will try it!

hirotakaster, 2016-11-30:
Now this use for server certification only I checked on mosquitto TLS server. 
And this can not use on Amazon IoT MQTT broker, because of AWS IoT server and client certification with private key processing is very heavy, TLS failed on malloc/calloc in RSA funcation.

hirotakaster, 2018-03-02:
Hi there,
MQTT with TLS is please check this repo. 
https://github.com/hirotakaster/MQTT-TLS

closed 2018-03-02 by hirotakaster