hirotakaster/MQTT #26: Not working with Elektron

ckrey, 2016-04-07:
I try to run a small GPS .ino on an Electron 2G.
The same .ino works great on a Photon.

any hints?

mosquitto server logs:
```
Apr  7 12:52:14 pi2 mosquitto[22768]: New connection from 176.83.137.98 on port 9001.
Apr  7 12:52:14 pi2 mosquitto[22768]: New client connected from 176.83.137.98 as 3a0021000c51343334363138 (c1, k15).
Apr  7 12:52:14 pi2 mosquitto[22768]: Sending CONNACK to 3a0021000c51343334363138 (0, 0)
Apr  7 12:52:36 pi2 mosquitto[22768]: Client 3a0021000c51343334363138 has exceeded timeout, disconnecting.
Apr  7 12:52:36 pi2 mosquitto[22768]: Socket error on client 3a0021000c51343334363138, disconnecting.
```

hirotakaster, 2016-04-07:
I think maybe client side tcp connection going down, server can't send packet to client. Please check and debug line 123 with Serial.println.

`
            while (!_client->available()) {
                unsigned long t = millis();
                if (t-lastInActivity > MQTT_KEEPALIVE*1000UL) {
                    _client->stop();
                    Serial.println("can't read from server packet");
                    return false;
                }
            }
            uint8_t llen;
            uint16_t len = readPacket(&llen);
            Serial.print("read packet length : ");
            Serial.println("len");

            if (len == 4 && buffer[3] == 0) {
                lastInActivity = millis();
                pingOutstanding = false;
                Serial.println("read packet OK");
                return true;
            }
        }
        Serial.println("read packet fail");
        _client->stop();
`

ckrey, 2016-04-07:
It seems to be a general TCP problem of the electron.

https://community.particle.io/t/solved-electron-tcpclient-weirdness/20771

I did install the beta firmware 0.5.0.-rc1 and had no problem using MQTT afterwards

closed 2016-04-07 by ckrey