hirotakaster/MQTT #28: Particle core: doesn't work with firmware 0.4.9 and above

natcl, 2016-05-23:
Hello, 
I tried this library and it doesn't seem to work with firmware 0.4.9 and above. Works fine with 0.4.7
Is this normal ?

hirotakaster, 2016-05-26:
Hi, I works well on default (0.5.0). Please tell me the problem.

hirotakaster, 2016-05-29:
Hi @natcl ,
Now I check Photon with 0.5.1 firmware working well. I will close this issue okay?

natcl, 2016-05-29:
Did you try with the Particle Core ? Never worked here... I tried with the
stock example.

hirotakaster, 2016-05-30:
Oh, I don't test on Spark Core. My spark core was broken, so I can't check on Spark core(now Spark Core can't buy on Particle). What's error happen? If you okay, please send bug-fixed pull request.

natcl, 2016-05-30:
Hello, it doesn't seem to connect at all, is there way I can enable more
debugging ?

hirotakaster, 2016-06-01:
If I check the following. I think it's may be problem at Core firmware or MQTT library.
1. Capture the packet on MQTT server side using tcpdump or wireshark for debug. Check incoming TCP packet and MQTT flow sequence on your Particle Core application and MQTT server.
2. use Serial.println debugging with Particle Dev application.

hirotakaster, 2016-10-01:
Hi, @natcl 
Now I check latest firmware 0.5.3 with my Core, MQTT library works well.

natcl, 2016-10-01:
Thanks, will test it also and report back.

natcl, 2016-10-14:
I just tested again and I have the same problem, works with firmware 0.4.7 but not above, here's the code I'm testing:

```C
// This #include statement was automatically added by the Particle IDE.
#include "MQTT/MQTT.h"

void callback(char* topic, byte* payload, unsigned int length);

/**
 * if want to use IP address,
 * byte server[] = { XXX,XXX,XXX,XXX };
 * MQTT client(server, 1883, callback);
 * want to use domain name,
 * MQTT client("www.sample.com", 1883, callback);
 **/
MQTT client("iot.eclipse.org", 1883, callback);

// recieve message
void callback(char* topic, byte* payload, unsigned int length) {
    char p[length + 1];
    memcpy(p, payload, length);
    p[length] = NULL;
    String message(p);

    if (message.equals("RED"))    
        RGB.color(255, 0, 0);
    else if (message.equals("GREEN"))    
        RGB.color(0, 255, 0);
    else if (message.equals("BLUE"))    
        RGB.color(0, 0, 255);
    else    
        RGB.color(255, 255, 255);
    delay(1000);
}

void setup() {
    RGB.control(true);
    
    // connect to the server
    client.connect("mf-core-client");

    // publish/subscribe
    if (client.isConnected()) {
        client.publish("mfAnnounce","hello world from Core !");
        client.subscribe("mfRGB");
    }
}

void loop() {
    if (client.isConnected())
        client.loop();
}

```

hirotakaster, 2016-10-14:
Hi @natcl ,
I checked lastest firmware version is 0.5.3.  0.4.7  is old firmware version.

natcl, 2016-10-14:
Yes, I know 0.4.7 is old but the code I sent only works with this on the
core, not with 0.5.3

hirotakaster, 2016-10-14:
Okay, But why do you have to use version 0.4.7? I think latest firmware is good.

natcl, 2016-10-14:
I don't want to use 0.4.7, but the example I sent doesn't work with 0.5.3,
it doesn't connect to the broker at all.

hirotakaster, 2016-10-14:
My environment is here, working well.
Server : mosquitto version 1.4.10, MQTT v3.1 broker.
Client : Particle Core firmware version 0.5.3. MQTT is 0.4.1.
Your MQTT server and other MQTT library(example...mosquitto_pub/sub) is works well?

osmarmorales, 2016-12-03:
Hi, I was using this library on a photon with firmware 5.2, yesterday update to version 6.0 stable and stop working the callback, the broker receives the message sent by the photon, but if I send a message to the photon this does not detect it, this only happens to with the photon that update, the photon in version 5.2 is working well. any ideas?

hirotakaster, 2016-12-04:
Hi @osmarmorales , My Particle Photon firmware 0.6.0 with MQTT library work well. Please show your code.

osmarmorales, 2016-12-05:
Hi @hirotakaster here is the code, As I said the code was working normally, Just update the code and along with this it updated the firmware of the photon:

`// This #include statement was automatically added by the Particle IDE.
#include "MQTT/MQTT.h"

// This #include statement was automatically added by the Particle IDE.
#include "Adafruit_DHT/Adafruit_DHT.h"

// This #include statement was automatically added by the Particle IDE.
#include "Adafruit_SSD1306/Adafruit_SSD1306.h"

#define DHTPIN 4
#define DHTTYPE DHT22
#define DEVICE_ID "Equip1"

DHT dht(DHTPIN, DHTTYPE);

int led = D7;

void callback(char* topic, byte* payload, unsigned int length);
byte host[] = {172, 16, 25, 100};
MQTT client(host, 1883, callback);

String celsius;
SYSTEM_MODE(SEMI_AUTOMATIC);
SYSTEM_THREAD(ENABLED);
STARTUP(WiFi.selectAntenna(ANT_EXTERNAL));

void setup() 
{
  WiFi.on();
  WiFi.connect();    
  Serial.begin(9600);
  dht.begin();
  pinMode(D3, OUTPUT);
  pinMode(D5, OUTPUT);
  pinMode(D6, OUTPUT);
  pinMode(D7, OUTPUT);
  digitalWrite(D3, LOW);
  digitalWrite(D5, HIGH);

 client.connect(System.deviceID());

 
 if (client.isConnected()) 
 {
        client.subscribe("RespSolicitud/"DEVICE_ID);
        client.subscribe("Settemp/"DEVICE_ID);
        client.subscribe("ONOFF/"DEVICE_ID);
        client.subscribe("Compresor");
        client.subscribe("Costos/"DEVICE_ID);
        client.subscribe("Reset");
        client.subscribe("Cloud");
        client.publish("Primera/"DEVICE_ID,status);
       
 }
 
}

void loop(void) {

 celsius = dht.getTempCelcius();
if((((celsius <= 10) && lastTemp > 15)) || celsius > 38) // This is what i update
    {
        celsius = lastTemp;
    }
    lastTemp = celsius;
 client.publish("Temperatura/"DEVICE_ID, celsius);

}

void callback(char* topic, byte* payload, unsigned int length) 
{
    char p[length + 1];
    memcpy(p, payload, length);
    p[length] = NULL;
    String message(p);
    String destino(topic);
    String w;
    int i = 0;
    
    String cmd[] = {"RespSolicitud/"DEVICE_ID, "Settemp/"DEVICE_ID, "ONOFF/"DEVICE_ID, "presor", "Costos/"DEVICE_ID, "Reset", "Cloud"};
    
    for (i = 0; i <= sizeof(cmd)/sizeof(cmd[0]); i++)
  {
    if (destino.equals(cmd[i]))
    break;
  }
    
    
    switch(i)
    {
        case 0:
        if (message.equals("2"))
    {
       solcomp=1;
       comp=0;
    }
    if (message.equals("1"))
    {
        digitalWrite(led, LOW);
        bp=0;
    }
    break;
    
    case 1:
    bandera=message.toFloat();
    break;
    
    case 2:
    primer=2;
    break;
    
    case 3:
    ultimocomp = message;
    break;
    
    case 4:
     consumo = 0;
    break;
    
    case 5:
    if(message.equals(DEVICE_ID))
    {
    System.reset();
    }
    if(message.equals("2"))
    {
    System.reset();
    }
    
    break;
    
    case 6:
    if(message.equals("ONLINE"))
    {
        Particle.connect();
    }
    if(message.equals("OFFLINE"))
    {
        Particle.disconnect();
    }
    break;  

    default: 
      Serial.println("Bad Command");
    
    } 
}`

hirotakaster, 2016-12-06:
Hi @osmarmorales , thank you. I can success when comment out SEMI_AUTOMATIC mode. Do you need SEMI_AUTOMATIC mode? That mode is needed check WiFi connection on yourself.

hirotakaster, 2016-12-08:
Hi, all. I will close this issue in 2-3 days.

osmarmorales, 2016-12-09:
Hi, sorry I was out of town, I solved the problem, it was in the loop area, specifically client.loop, Thank you very much for your help, I appreciate it.

closed 2016-12-09 by hirotakaster