hirotakaster/MQTT #30: Not really an issue

duffo64, 2016-07-19:
Hello @hirotakaster ,
first of all, great work with this client.
It's not clear to me if is possible to reuse a declared client changing ip address. 

Use case:
I have two independent mqtt clusters, and would to switch to the second in case of failure of the first.

True, I could have two client objects (one for cluster), but the program would be a lot less linear.
Thank you for your time and efforts

Claudio

hirotakaster, 2016-07-19:
I think that's system architecture question. You means like following.
[Photon(MQTT Client)]  <----> [MQTT server 1(health: good): IP Address : X.X.X.X]
                                                    [MQTT server 2(health: good): IP Address : Y.Y.Y.Y]
when MQTT cluster server1 down because of some problems, MQTT client re-connect to the other MQTT server.
[Photon(MQTT Client)]  <-|           [MQTT server 1(health: down): IP Address : X.X.X.X]
                                            |---->  [MQTT server 2(health: good): IP Address : Y.Y.Y.Y]
maybe you can do this using by isConnected() method or tow MQTT object. But if I build same system, I will do following.

[Photon(MQTT Client)]  <-----> LVS(HA and keepalived or heatbeat): Virtual IP : A.A.A.A)
                                                      [MQTT server 1(health: good): IP Address : X.X.X.X]
                                                      [MQTT server 2(health: good): IP Address : Y.Y.Y.Y]

LVS check the MQTT server 1,2... health status and switch the connect packet from the client, MQTT client only always connect to the LVS virtual IP A.A.A.A.
There is good MQTT client no need to have two MQTT client object(only implements re-connect) and LVS manage some MQTT servers(1,2,3...) health-check and more(load-balance) very easily.

duffo64, 2016-07-19:
Yes, I know thanks. I have already two distinct HAproxies, each of them is balancing 3 mqtt brokers. I simply would avoid clustering HAproxy, since I know that one day the whole system will fail, so I would like to manage two instances in code. Moreover, an HAproxy cluster is more difficult  to configure, and this system will work with almost untrained personnel, so I would keep things simple.

Thank you, nonetheless.

Claudio

closed 2016-08-22 by hirotakaster