hirotakaster/MQTT #31: Compatibility with Particle Electron

iamsalman, 2016-08-09:
Hi. is it also compatible with Particle Electron?

developsoftware, 2016-08-10:
It works perfectly on my electron.

iamsalman, 2016-08-10:
I am trying to connect electron to IBM Bluemix. I am using mqtttest.ino in your examples folder with my custom parameters for server and IOT_Client, username and password for "client.connect" function. The device becomes online for around 30 seconds and it remains stuck in the "client.connect" function of the library. The function returns 'false'(I checked on console) and the device goes offline after 30 seconds. And just to inform you I am quite a newbie so sorry if I am missing something quite basic. Can you please figure out from this description what might be the problem? Thank you!

perotom, 2016-08-10:
please check if your device is connected to the internet (check in the particle cloud, is it flashing cyan?)
after that you can try to use an open mqtt server (you can find on the eclipse website a testing server)

iamsalman, 2016-08-11:
My device is connected and breathes cyan. I have tried on local mosquitto server and these are the logs:
---New connection from ***.**.***.** on port 1883.
---New client connected from ***.**.***.** as testClient (c1, k15, uparticle).
---Sending CONNACK to testClient (0)
---Client testClient has exceeded timeout, disconnecting.

The disconnection happens after about 30 seconds of connection

perotom, 2016-08-11:
that´s somehow strange. have you used the example sketch?

iamsalman, 2016-08-12:
Problem is solved now. Issue was that firmware of Electron device needed to be upgraded. Although i bought it new but still update was required. Thanks everyone for ur help.
For all others in future, always upgrade ur device before everything else!

hirotakaster, 2016-08-22:
thank  you for your help.

closed 2016-08-22 by hirotakaster