hirotakaster/MQTT #33: initializing argument 2 of 'bool MQTT::publish(const char*, const uint8_t*, unsigned int)' [-fpermissive]

rayterrill, 2016-08-21:
Having a heck of a time trying to use the MQTT library in Particle with integer values coming off of a moisture sensor.

Veryfing code using this library in Particle Build fails with this error:
```
MQTT/MQTT.h:140:10: error:   initializing argument 2 of 'bool MQTT::publish(const char*, const uint8_t*, unsigned int)' [-fpermissive]
     bool publish(const char *, const uint8_t *, unsigned int)

Here's the code that's giving me the problems:
`  if (client.isConnected()) {
     int sensorValue = 0;  // variable to store the value coming from the sensor
     
     // read the value from the sensor:
     Serial.println("Reading sensor value...");
     sensorValue = analogRead(sensorPin);
     
     Serial.println(sensorValue);
     Serial.println("Attempting to publish a reading...");
     client.publish("rayterrill/feeds/moisture",(uint8_t)sensorValue,4);
     //client.subscribe("/inTopic");t

Trying to cast the integer value to uint8_t, but that gives the error above. Switching it to (uint8_t*) clears the issue, but I get blank or garbage data in the Adafruit IO dashboard feed I'm using, depending what I have the plength set to.

Am I just doing something totally wrong here?

hirotakaster, 2016-08-22:
maybe like following.

    // client.publish("rayterrill/feeds/moisture",(uint8_t)sensorValue,4);
    client.publish("rayterrill/feeds/moisture",(uint8_t *)&sensorValue, sizeof(sensorValue));

rayterrill, 2016-08-22:
I'm still getting garbage even with that code.

I looked at the Adafruit MQTT library, and they've got a publish declaration that allow for an int value, as in this:
`bool Adafruit_MQTT_Publish::publish(int32_t i) {
  char payload[12];
  ltoa(i, payload, 10);
  return mqtt->publish(topic, payload, qos);
}
`

I incorporated some similar char coercing logic into my publish, and I'm indeed now able to publish data using this libraries char publish declaration. I just can't get it to work at all with the unit8_t declaration.

hirotakaster, 2016-08-22:
Do you want to use char or int type in publish? If you want to use char, you can do like this.

<pre>
char publishData[120];
sprintf(publishData, "%d", 10);
client.publish("rayterrill/feeds/moisture", publishData, strlen(publishData));
</pre>

or If you want to use int type. of course you should get published data with int type.

<pre>
  client.publish("rayterrill/feeds/moisture",(uint8_t *)&sensorValue, sizeof(sensorValue));
</pre>

rayterrill, 2016-08-22:
I tried it with:
`client.publish("rayterrill/feeds/moisture",(uint8_t *)&sensorValue, sizeof(sensorValue));`

Still getting garbage data into Arduino IO. It might just be Arduino IO, not sure.

I wanted to post as int, but char is also working.

hirotakaster, 2016-08-23:
here is my test code.

<pre>
// photon side
int sensorValue = 65; // character 'A'
void loop() {
    if (client.isConnected())
        client.loop();
    delay(1000);
    client.publish("sensor/val/", (uint8_t *)&sensorValue, sizeof(sensorValue));
    sensorValue++;
}
</pre>

<pre>
// mosquitto side
mosquitto_sub  -h [mosquitt_server_ipaddr] -t "sensor/val/"
A
B
C
D
E
F
...
</pre>

I check works well on my Photon/mosquitto command. Check your Arduino IO side source code.

hirotakaster, 2016-08-29:
Hi @rayterrill , can you work on Photon with MQTT? I will close this issue.

rayterrill, 2016-08-29:
Sure. Go ahead and close. Thanks for the help!

closed 2016-08-30 by hirotakaster