hirotakaster/MQTT #35: Please acknowledge PubSubClient

knolleary, 2016-09-19:
Hi,

it appears this library is based largely on my PubSubClient library.

However this library makes no reference that I can see of my library or the appropriate copyright. Please refer to the license I share my library under; you are free to use the code, you just have to acknowledge where it came from.

https://github.com/knolleary/pubsubclient/blob/master/LICENSE.txt 

> The above copyright notice and this permission notice shall be
included in all copies or substantial portions of the Software.

Nick

hirotakaster, 2016-09-19:
Hi, oh, It's a sorry.
I wrote your license on MQTT.h header file, but I forget write lisence file. 
Now I update lisence.txt file, please check this.

hirotakaster, 2016-09-23:
Hi, @knolleary 
Does it Okay? Can I close this?

knolleary, 2016-09-23:
Thanks for addressing!

closed 2016-09-23 by knolleary