hirotakaster/MQTT #40: Photon Disconnects and won't reconnect

cketcham, 2016-10-31:
Hi,

Thanks for the awesome library! I have some problems with my photons disconnecting from the broker and never connecting again. In your example you show that you should run `if (client.isConnected()) client.loop();` in `loop()`. I'm doing that but my photons lose their connection and `client.loop()` will never try to reconnect. I'm guessing that the photons loose their connection to the wireless router or something and then disconnect from the broker. In order to solve that problem I try to reconnect the client if `client.loop()` returns false. ie. something like `if (!client.loop()) client.connect()`

Do you know if there's something I should fix on my photons or mqtt broker to stop it from disconnecting? Maybe there should be a mode of the client which will try and keep it connected to the broker even if it disconnects for longer than the keepalive time after it sends MQTTPINGREQ? thanks.

I'm using particle photons with firmware 0.5.3, version 0.4.1 of your library, and mosquitto version 1.4.10 (build date 2016-10-25 21:29:55-0400).

hirotakaster, 2016-10-31:
Hi @cketcham 

I'm using this source code.
<pre>
#include "MQTT/MQTT.h"
void callback(char* topic, byte* payload, unsigned int length);
MQTT client("MQTT Broker", 1883, callback);

// recieve message
void callback(char* topic, byte* payload, unsigned int length) {
    char p[length + 1];
    memcpy(p, payload, length);
    p[length] = NULL;
    String message(p);

    if (message.equals("RED"))    
        RGB.color(255, 0, 0);
    else if (message.equals("GREEN"))    
        RGB.color(0, 255, 0);
    else if (message.equals("BLUE"))    
        RGB.color(0, 0, 255);
    else    
        RGB.color(255, 255, 255);
}

void setup() {
    RGB.control(true);
    Serial.begin(9600);
}

void loop() {
    if (client.isConnected()) {
        Serial.println("mqtt client loop");
        client.loop();
    } else {
        Serial.println("mqtt connect loop");
        client.connect("sparkclient");
        if (client.isConnected()) {
            client.subscribe("inTopic");
        }
    }
    delay(1000);
}
</pre>

test case is following.
1. start mosquitto server.
2. start Photon to PC and connecting WiFi AP.
3. check serial console output with Particle Dev app 
4. send mosquitto_pub message and change LED.
`mosquitto_pub  -h mqttbroker -t "inTopic" -m "RED"`
5. stop mosquitto server, check the MQTT connection is lost.
6. start mosquitto server, check the MQTT isConnected() == true.
7. send mosquitto_pub message and change LED.
`mosquitto_pub  -h mqttbroker -t "inTopic" -m "BLUE"`

cketcham, 2016-10-31:
Ok, that's pretty much what I'm doing and it works. I just wanted to make sure you weren't doing something different since the examples don't show the code to reconnect. thanks.

hirotakaster, 2016-11-01:
Hi @cketcham , that's good. close this issue?

closed 2016-11-01 by cketcham