hirotakaster/MQTT #42: Size of message to be sent

Biribu, 2016-12-04:
I am trying to send some data to my mqtt broker and I have found that as soon as the message contains a 0, the rest of it is not paid attention.

For example: 
char a[10];
a[0] = 3;
a[1] = 2;
a[2] = 0;
a[3] = 4;

The message received in broker seems to be 32 and nothing else. 

I trying with photon 0.5.3

Is there a way to specify the length of the message so it can contains whatever I want?

hirotakaster, 2016-12-04:
hi, could you show your code? I did not get that.

Biribu, 2016-12-04:
Sure.

I have a char array declared as:
char frameRegister[30];

Then, inside a function (which connects to my mqtt server and sends a frame):

for(i = 0; i < 30; i++) {
      frameRegister[i] = 0;
    }
frameRegister[0] = mac[0];
    frameRegister[1] = 3;
    frameRegister[2] = 5;
    frameRegister[3] = 5;
    frameRegister[4] = 6;
    frameRegister[5] = 7;
    frameRegister[6] = 0;
    frameRegister[7] = 255;

client->connect("clientId");
  if (client->isConnected()) {
     client->publish("frame/service",frameRegister);
  }

On mqtt.fx client, I just get the message until 7, no 0 and no 255.

hirotakaster, 2016-12-04:
use like this,

    uint8_t frameRegister[30];
    ....
    client->publish("frame/service",frameRegister, sizeof(frameRegister), false, MQTT::QOS0,  NULL);

you have to tell frameRegister size to MQTT client.

hirotakaster, 2016-12-04:
or simple version is this
 
    uint8_t frameRegister[30];
    ...
    client->publish("frame/service",frameRegister, sizeof(frameRegister));

try this.

hirotakaster, 2016-12-06:
@Biribu close this issue?

Biribu, 2016-12-06:
Sure.
Thank you @hirotakaster , that resolved my problem.

closed 2016-12-06 by Biribu