hirotakaster/MQTT #43: Multiple Photons loose connection

BaudGit, 2016-12-06:
Hello and thanks for your MQTT library.

I´ve been testing your library on one Photon without any problem. I ran your example and I am able to switch the LED color by publishing from another MQTT user/client successfully.

Nevertheless, if I use the same code (of course changing the user and password) simultaneously on another photon, the first photon will loose its connection. In other words, only the last photon that connects to the server will keep its connection. All of these photons are connected through the same WiFi.

This is the code I´m testing, I just added a LED indicator to your example to see if the connection is alive:

```
#include "MQTT/MQTT.h"

int led = D7;

void callback(char* topic, byte* payload, unsigned int length);

/**
 * if want to use IP address,
 * byte server[] = { XXX,XXX,XXX,XXX };
 * MQTT client(server, 1883, callback);
 * want to use domain name,
 * MQTT client("www.sample.com", 1883, callback);
 **/
MQTT client("m13.cloudmqtt.com", #####, callback);

// recieve message
void callback(char* topic, byte* payload, unsigned int length) {
    char p[length + 1];
    memcpy(p, payload, length);
    p[length] = NULL;
    String message(p);

    if (message.equals("RED")){    
        RGB.color(255, 0, 0);
    }
    else if (message.equals("GREEN"))    
        RGB.color(0, 255, 0);
    else if (message.equals("BLUE"))    
        RGB.color(0, 0, 255);
    else    
        RGB.color(255, 255, 255);
    delay(1000);
}

void setup() {
    RGB.control(true);
    pinMode(led, OUTPUT);
    
    // connect to the server
        client.connect("m13.cloudmqtt.com","User1","Password1");

    // publish/subscribe
    if (client.isConnected()) {
        client.publish("TopicTest","hello world2");
        client.subscribe("TopicTest");
    }
}

void loop() {
    if (client.isConnected()){
        client.loop();
        digitalWrite(led,HIGH);
    }else{
        digitalWrite(led,LOW);   
    }
}
```

I thought it was an issue concerning the wifi connection so I added this code to restart the connection if it is lost:

```
void loop() {
    if (client.isConnected()){
        client.loop();
        digitalWrite(led,HIGH);
    }else{
        digitalWrite(led,LOW);   
        client.connect("m13.cloudmqtt.com","User1","Password1");
        client.subscribe("TopicTest");
    }
}
```

But still the last photon to connect to the server is the one that keeps the connection. I observe the LEDs switching on and of in the inverse signal of each other each time one of them opens the connection with the server.

I´ve tested this situation with up to 3 photons simultaneously and only one is able to connect to the server at each time.

The mqtt broker lives in mqttcloud with a 100 connections, 100 Kbit/s capability.

One might suspect on the MQTT broker but I´ve checked with another non-Photon clients and they are able to keep the connection without making the first photon loose its connection.

Am I using your library in a wrong way?

Thank you in advance.

BaudGit, 2016-12-06:
It seems that the issue is that the library declares the same Client-ID to the broker, an thus with many instances of the library the broker rejects the connections that use the same Client-ID.

¿How can I change the Client-ID?

hirotakaster, 2016-12-07:
Your problem is dependent on MQTT broker spec.
Normally, MQTT broker(mosquitto...etc) will close first connection when same client id connection coming.  
You can change the client id on connect method 1st parameter = id.

    bool MQTT::connect(const char *id)

BaudGit, 2016-12-07:
Great! That fixed it. Thank you!

closed 2016-12-07 by BaudGit