hirotakaster/MQTT #46: I can't get device name while connecting to MQTT for some reason.

peterwilli, 2016-12-22:
I've found out about this and explained it further here: https://community.particle.io/t/photon-device-name-not-returning-anymore/28088/4

I'm using the following code to connect to mqtt:

```
void callback(char* topic, byte* payload, unsigned int length) {
    char p[length + 1];
    memcpy(p, payload, length);
    p[length] = NULL;
    String message(p);

    Particle.publish("MQTT: " + message);
}
MQTT client("m13.cloudmqtt.com", 18561, callback, 512);

unsigned long lastConnection = 0;
void loop() {
    if (client.isConnected()) {
        client.loop();
    } else {
        if(lastConnection == 0 || (millis() - lastConnection) > 5000) {
            lastConnection = millis();
            Particle.publish("MQTT Connection loop");
            client.connect("m13.cloudmqtt.com", "xxxx", "xxxxx");
            return;
        }
    }
}
```

For some reason, when I am running this code in the loop Particle's `Particle.publish("spark/device/name");` does not reply anymore.

hirotakaster, 2016-12-26:
Hi @peterwilli ,
Particle publish method is based CoAP message(UDP), your problem does not matter with MQTT library. Maybe firmware bug or others. You should be better update your issue to Particle firmware github.

AndrewWeiss, 2017-07-14:
Peter,

I don't have the answer but I have a suspicion. In running into a different issue having to do with particle and also using the library. Which so far has been a pretty well written library. I think the issue has to do with the need to call Particle.process() when you have system threading enabled. There's is a while true loop in the mqtt connect that I think needs particle.process in it to function properly when the connection blocks for too long. Work in progress...

Andy