hirotakaster/MQTT #48: Proposal for addition of custom keep-alive interval

imaustink, 2017-02-15:
I am interested in adding a custom keep alive interval to this library. A recent project requires a near real-time notification of disconnected client. A keep-alive interval of 1 second should be fine.

I imagine something like:
```c++
#include MQTT.h

MQTT client(server, 1883, callback);

void setup() {
    // Optional second arg of the connect method is an int representing number of seconds
    client.connect("ClientName", 1);
    if (client.isConnected()) client.publish("test","Hello World");
}

void loop() {
    if (client.isConnected()) client.loop();
}
```

hirotakaster, 2017-02-16:
hi @imaustink 
thank you for your pull request!! I will check your commit and merge later.

hirotakaster, 2017-02-17:
okay, now I merge all your commit.
Arduino compatible code parts are deleted but it maybe no problem:)

imaustink, 2017-02-17:
Thank you kind sir for the timely response!

I know the Arduino stuff isn't an problem but I feel that for maintainability moving forward it would be good to have a better separation for platforms and a cleaner codebase for each :)

imaustink, 2017-02-17:
Thank you kind sir for the timely response!

I know the Arduino stuff isn't an problem but I feel that for maintainability moving forward it would be good to have a better separation for platforms and a cleaner codebase for them as well :)

Thanks!

closed 2017-02-17 by imaustink

hirotakaster, 2017-02-17:
Yes, already update to Particle library version 0.4.6. Please check it and close this issue.