hirotakaster/MQTT #50: Support for changing server address after constructing object

foodbag, 2017-03-04:
Would it be possible to support changing the server address after constructing the mqtt object? This would be useful if one wants to change the mqtt broker address without refreshing.

hirotakaster, 2017-03-07:
hi @foodbag ,
okay, I will update initialize method to public. then disconnect/connect like this.
```
mqtt.publish("topic/", "hoge"); 
mqtt.disconnect();
mqtt.initialize("example.com", NULL, 1883, callback, MQTT_MAX_PACKET_SIZE);
mqtt.publish("topic/", "hoge");
```

foodbag, 2017-03-07:
That would be great. Thanks!

foodbag, 2017-03-10:
Using the initialize method to update the broker address may cause a memory leak if called multiple times before resetting. It might be safer to update just the domain/ip and port rather than reinitializing.

PR #52 (2017-03-13): Added setBroker method to update domain or ip and port. 
Changed initialize method back to private. this commit fixes #50
commit 010e8b47 (2017-03-13): Changed initialize method back to private. Added setBroker method to
update domain or ip and port. this commit fixes #50
closed 2017-03-13 by hirotakaster