hirotakaster/MQTT #51: Missing header guards in MQTT.h

alignan, 2017-03-04:
To avoid errors like `error: previous definition of 'class MQTT'`

hirotakaster, 2017-03-07:
I don't know what's happen on only this error line.

ScruffR, 2017-04-11:
@hirotakaster, the common practice for all headers should be to guard against multiple inclusion.
This is either done as
```
#ifndef _YOUR_LIBRARY_H_
#define _YOUR_LIBRARY_H_
// here follows the code
#endif
```
or via 
```
#pragma once
// here follows your code
```

hirotakaster, 2017-04-12:
Hi @ScruffR 
I add ifdef to header file, but I can't use MQTT my lib on WebIDE and local Particle Dev!! compile message is "MQTT.h: No such file or directory".
```
#include <MQTT.h>
#include "MQTT.h"
#include "MQTT/MQTT.h"
```
all patter is fail "MQTT.h: No such file or directory" now. 
I don't know why this happen, but I think Particle community lib is broken now.

ScruffR, 2017-04-12:
That is due to an issue on the Particle side.
https://github.com/spark/particle-dev-libraries/issues/26

You need to remove the nested `MQTT` folder with the header stubs

hirotakaster, 2017-04-12:
@ScruffR 
Okay, thank you. I can compile on "Particle Dev" local pc. 
But still can't compile on WebIDE.

closed 2017-05-13 by alignan