hirotakaster/MQTT #58: Build fails because of missing application.h

alignan, 2017-05-13:
Hello!
After removing `#include "application.h"` from `MQTT.cpp`, the application fails to compile:

````
In member function 'bool MQTT::connect(const char*, const char*, const char*, const char*, MQTT::EMQTT_QOS, uint8_t, const char*)':
MQTT/MQTT.cpp:156:55: error: 'millis' was not declared in this scope
             lastInActivity = lastOutActivity = millis();
                                                       ^
In member function 'bool MQTT::loop()':
MQTT/MQTT.cpp:237:34: error: 'millis' was not declared in this scope
         unsigned long t = millis();
                                  ^
In member function 'bool MQTT::write(uint8_t, uint8_t*, uint16_t)':
MQTT/MQTT.cpp:419:30: error: 'millis' was not declared in this scope
     lastOutActivity = millis();
                              ^
In member function 'void MQTT::disconnect()':
MQTT/MQTT.cpp:467:47: error: 'millis' was not declared in this scope
     lastInActivity = lastOutActivity = millis();
````

This is a [known topic](https://community.particle.io/t/httpclient-library-wont-compile-on-photon/12461), as Particle build requires `application.h`.  Including the header fix the problem

hirotakaster, 2017-05-13:
Hi @alignan,
`application.h` is already included "MQTT.h" in latest version, did not be included in `MQTT.cpp` now.

alignan, 2017-05-13:
I don't see the include in the `master` branch

hirotakaster, 2017-05-13:
https://github.com/hirotakaster/MQTT/blob/master/firmware/MQTT.h

hirotakaster, 2017-05-13:
@alignan 
Okay, Now I checked on Particle WebIDE and local IDE, in that's environment everything okay.
Maybe I forget the update to the github.

hirotakaster, 2017-05-13:
now fixed header.

closed 2017-05-13 by hirotakaster