hirotakaster/MQTT #63: Publish rate

eushay, 2017-06-21:
Hello, 
Recently I used an accelerometer with photon running at 100 Hz. Every output value was sent to MQTT broker to see the result. However when I compared the result from MQTT broker with serial monitor many entries were missed. The issue might be that the publish rate is not fast enough while using this library. Is there any way that the library can be tweaked to increase the publish rate.

hirotakaster, 2017-06-21:
Hi, @eushay 
100Hz is too fast for Photon network stack on WiFi, not a MQTT library. If you try to use just a TCPClient, the results will be the same. You should adjust the speed. I think maybe max speed is 10-20Hz on Photon network stack.

eushay, 2017-06-21:
Thanks, didnt knew about that.

closed 2017-06-21 by eushay