hirotakaster/MQTT #64: Cannot connect with username/password to mosquitto 1.3.5

mikewop, 2017-06-24:
I am using a Particle Photon connecting to a mosquitto broker on the same subnet.

If mosquitto is set to allow anonymous connections (allow_anonymous = true) then the connection works.
If mosquitto is set to require username/password (allow_anonymous = false) then an error message is received: 
`[DEBUG] MQTT  Connect fail. code = [5]`

This username/password works fine when using mosquitto_pub from another computer.
I am sure the username and password are correct, mosquitto would send back a return code of 4 for incorrect username password.
Return code of 5 seems the username/password is not even passed into mosquitto.

I am using the MQTT library from the Particle WEB IDE (build.particle.io).
`    // connect to the server
    client.connect("photon_04","hauser", "mosq_hapswd");`

TypQxQ, 2017-06-25:
I have the same problem. I get the following in the Mosquitto log:
```
1498397931: Socket error on client <unknown>, disconnecting.
1498397949: Received PINGREQ from home-assistant-1
```
I'm running Mosquitto 1.4.12. 
Photon with Particle 0.6.2 and MQTT 0.4.20 from the Web IDE.

When changing "allow_anonymous false" to "true" it just works without any changes.
Code unchanged. Only changes where I upgraded Particle from 0.6.0 to 0.6.2 and MQTT from an older version that wasn't compatible with 0.6.2.

hirotakaster, 2017-06-26:
Hi @mikewop , @TypQxQ 
now I fixed this and update to 0.4.21. you can use on Web IDE.

mikewop, 2017-06-26:
@hirotakaster Thank you! It works great now.

closed 2017-06-26 by mikewop