hirotakaster/MQTT #65: Does QOS2 also queue messages if connection is gone while publishing the QOS2 message?

kenvernaillenvub, 2017-08-10:
When I publish a certain message while internet is down, it should be queued and sent later on. Is this already implemented in QOS2? If it is QOS2, will this keep trying to send it? Because the first line in MQTT::publish is `if (isConnected()) {` and I think it will automatically return false and don't send it. What is the best way to make sure I don't loose any QOS2 messages?

hirotakaster, 2017-11-06:
@kenvernaillenvub 
Hi, this lib don't store message on QoS=2. Instead of store the message, you can use the QoS2 message callback function(addQosCallback) for check the message really sent or not on your application.

mkesselaers, 2018-11-23:
Hi,

I have a similar request.
How would you tackle this?
Using the callback would mean that you would need to build a dict if you have multiple QOS2-messages that need to be pushed.

I'm quite new to MQTT so if you could provide an example, I would be forever grateful

Thanks,
Maarten