hirotakaster/MQTT #66: Mqtt Fails to connect

fcabrera23, 2017-08-13:
Sometimes the client fails to connect for some time. I have to reset Electron so that the entire connection restarts ( i've checked and the internet connectivity is ok and even send a mqtt.disconnect() before trying a connection)

fcabrera23, 2017-08-14:
Basically this is my loop 

if (Cellular.ready())
    {
        digitalWrite(ledGPRS,0);
        attempts = 0;
        if (mqttClient.isConnected()) {
            mqttClient.loop();
        } else {
            Serial.println("Connect to the MQTT server!!");
            mqttClient.connect(clientId,username,password);
            if (mqttClient.isConnected()) {
                Serial.println("Connected");
                mqttClient.addQosCallback(qoscallback);
                mqttClient.publish("reconnections", celu.imei);
            }
        }
    }
    else
    {
        digitalWrite(ledGPRS,1);
        if(attempts > 10){ // Si ya probe 10 (mas de 50 minutos intentando la conexion) veces la conexion y no tuve exito
            System.reset();
        }
        
        if ((millis() - lastConnect) >= CONNECT_TIMEOUT)
        {
            if(bomba_encendida == APAGADO) // Si la bomab esta desconectada
            {
                Cellular.disconnect();
                Cellular.off();
                delay(5000);
                debug < 1 ? debug=0:Serial.println("Intento conectar modem celular.");
                Cellular.on();
                Cellular.connect();
                lastConnect = millis();
                attempts++;
            }
        }
    }

Sometimes it connects, but if reset it, when trying again , won't connect again

imomo0, 2017-09-27:
I also have this issue running the example mqtttest.ino on the Photon. Could the issue be on the TCPClient lib?

hirotakaster, 2017-11-06:
@imomo0 
Update and try new firmware version.

leafgarden, 2021-05-01:
I had a similar issue and had to implement a write timeout, that seemed to solve the problem. I based my changes off an older version so would need to re-port it before doing a pull request. If this is useful to someone I can try to port it sooner so they can test my changes. If someone else wants to do the port here are some of the changes I did:

I replaced all _client.write(...) with this:

<pre>
bool MQTT::writeBuffer(const uint8_t *b, size_t s) {
    if ( timeout < 0 ) {
        return s != _client.write(b,s);
    }
    size_t ptr = 0;
    system_tick_t now = millis();
    system_tick_t start = now;
    _client.clearWriteError();
    while (ptr < s && (now-start) <= (system_tick_t)timeout) {
        size_t rlen = _client.write(&b[ptr],s-ptr,timeout);
        if (_client.getWriteError() != 0 ) {
            break;
        }
        ptr +=rlen;
        now = millis();
    }
    if ( ptr != s ) {
        _client.stop();
        return false;
    }
    return true;
}
</pre>

This now returns true/false so the MQTT::write(..) needs to check for true instead of comparing the length.

I added a setter for timeout and defaulted it to -1 so the behavior is not changed if you don't set a timeout.

I allow for a timeout of 0 which is truly non blocking but don't think it is useful, probably need to use a more complex Async stye library if you want that.