hirotakaster/MQTT #67: Cannot flash electron OTA if MQTT clinet is connected.

kostbill, 2017-10-25:
If I run the mqtttest I cannot OTA my electron.
If I don't connect to MQTT broker then I can program it.

This is in the electron. In the photon it is working fine.

hirotakaster, 2017-11-06:
@kostbill 
Please tell me the details or source code. I don't have your problem.

kostbill, 2017-11-06:
Hi, thanks for answering me.

Later today I will bring up the example and paste the code here.

CameronTurner, 2018-03-09:
I have found a similar issue. The workaround was to use client.disconnect() triggered via the cloud, then to run the OTA update. 

But this is a poor workaround and implies something is wrong elsewhere. I'm not sure, but some assumptions (given I'm running in threaded mode).

Some possibilities could be:
1. Code is blocking the network stack to the OTA function can not connect (or can't fully complete a full load of data) (I have noticed other cloud API calls are slower then the MQTT library is running)
2. The ping time is too short that when it runs, it is using up too many resources for the device to be able to run a OTA update and MQTT transmission at the same time. Either memory or shared resource related?
3. The MQTT comms coming back from the broker hold up something in the TCP/comms stack that results in the Particle connection being lost or timing out because it is sharing too much throughput?

Not sure!

CameronTurner, 2018-03-09:
@kostbill 

Did you find a resolution? or cause?

hirotakaster, 2018-03-09:
Hi @CameronTurner ,

1-3 that maybe firmware & tcpclient api problem. this library network stack is using tcpclient on particle. check the tcpclient&OTA or firmware source code with your non-blocking simple source code.