hirotakaster/MQTT #68: Give an example of publish with retained bit set

raenrfm, 2017-10-27:
I'm trying to send payloads that I want retained, but I cannot figure out the syntax of the publish statement to allow that.  I'm using QOS0 so don't care about that.

hirotakaster, 2017-11-06:
@raenrfm 
you use this function.
`
    bool publish(const char *topic, const uint8_t *payload, unsigned int plength, bool retain);
`

raenrfm, 2017-11-06:
This is the modified callback function correct?  But how is it used in the sketch?  Is it just the client.publish(topic,payload); command?  Do I have to pass anything else in this command or is modifying the callback function enough?

hirotakaster, 2017-11-06:
@raenrfm 
If you want to use retain flag use this, but I don't know what you want to do.

raenrfm, 2017-11-06:
But how is it used in the sketch?  Can you just show me a sketch that publishes with the retained bit set to true?  I want my MQTT server to retain values even if they have been read by one device.  I want the value to stay in the server until a new value replaces it.  That is what the retained message is for.

hirotakaster, 2017-11-06:
yes, you set "retain = true" like this `publish("/topic", "hoge", 4, true); `. 
This library is OSS(MIT), so you can see source code by yourself, and if you need more function or implementation please send a diff/pull request.

hellbe, 2017-12-01:
@hirotakaster I've studied the source code but can't get this to work either, tried casting input values to different types without success. Particle compile service returns this:

`call of overloaded 'publish(const char[18], const char[3], int, bool)' is ambiguous`

hellbe, 2017-12-01:
Use case for me is setting up an remote switch integration with Home Assistant ([docs](https://home-assistant.io/components/light.mqtt/)). The retain flag is used to feed back status after switching on/off.

Example `topic="bedroom/light/status", payload="ON"`

hellbe, 2017-12-01:
Darn.. the curse of posting questions. After looking one more time at the source I noticed the small type difference of the payload, this works for me now:

`client.publish("bedroom/light/status", (uint8_t*) "ON", 2, true);`

Ping @raenrfm in case you haven't solved it

raenrfm, 2017-12-02:
Ok so the addition of the (uint8_t*) is what solved the issue?

hellbe, 2017-12-02:
Yep @raenrfm

raenrfm, 2017-12-06:
Would this apply if your payloads are just unsigned integers?

raenrfm, 2017-12-06:
Sorry what I meant was that I'm using this on a particle photon and have
the payloads set as unsigned int's

<object width="290" height="130"><param name="movie" value="
http://www.wunderground.com/swf/pws_mini_rf_nc.swf?station=ITHUNDER5&freq=&units=metric&lang=EN"
/><embed src="
http://www.wunderground.com/swf/pws_mini_rf_nc.swf?station=ITHUNDER5&freq=&units=metric&lang=EN"
type="application/x-shockwave-flash" width="290" height="130" /></object>

raenrfm, 2017-12-06:
I really wish he created an example of using the retained feature.  I'm
still having trouble getting this to work.

hellbe, 2017-12-06:
I think you're ok as long as you're not overflowing it with a high number, uint8_t is short for an unsigned **8 bit** integer (i.e. 0-255). But I'm kind of out of my depth here :)

hellbe, 2017-12-06:
Check out this thread for convering high ints to uint8_t arrays: https://forum.arduino.cc/index.php?topic=361659.0

raenrfm, 2017-12-06:
yeah, it's working without the retained setup, but I don't know what
callback function to use to publish a retained payload.

raenrfm, 2017-12-06:
He has three example ino files but none of them show the use of the
retained bit.

hellbe, 2017-12-06:
You might need to clarify your use case more. The original question was just about publishing with retained flag but now you're talking about callbacks? 

I provided an example how to publish with a retained bit [in this comment](https://github.com/hirotakaster/MQTT/issues/68#issuecomment-348635615). If you have a high int as the payload you want to suggest the recommendation from the arduino forum thread I linked before is to convert your int to a string and then back to int on the receiving end.

raenrfm, 2017-12-06:
If you don't use the correct callback function you can't use the retained
bit capability, from what I understand.

raenrfm, 2017-12-06:
Here is the callback function used in his example:

void callback(char* topic, byte* payload, unsigned int length);

if you attempt to use this with the retained bit option in your publish
statement it blows up.

raenrfm, 2017-12-06:
I'm converting floats to strings then publishing.  I'm thinking that's the
difference between your case and mine.

dakl, 2019-02-12:
FYI, I added a method like so:

```
bool MQTT::publish(const char* topic, const char* payload, bool retain) {
    return publish(topic, (uint8_t*)payload, strlen(payload), retain, QOS0, NULL);
}
```

With this, I just do `client.publish("someTopic", "ON", true);` to send a message with the retain flag.

I'm happy to provide a PR if it's useful to the community @hirotakaster

hirotakaster, 2019-02-13:
Hi @dakl ,
Thank you!! Of course, PR welcome:)

dakl, 2019-02-13:
#77 should fix this. @raenrfm @hirotakaster