hirotakaster/MQTT #69: Disconnects on Particle Core

natcl, 2017-11-01:
With the mqtttest example, I cannot get a stable connexion on the Particle Core.  I tried with a remote broker and a local one with the same results.  After a few hours it stops responding and on the broker side I get a Socket Error from the client.
Any ideas what could be causing this ?

natcl, 2017-11-02:
Update:
Seems to be related to the connection keepalive, here's what I get on my broker:
```
client core-client has exceeded timeout, disconnecting
Socket error on client mf-core-client
```

In my loop I have this code which I assume should reconnect:

```c++
void loop() {
    if (client.isConnected()) {
        client.loop();
    } else {
        delay(1000);
        client.connect("mf-core-client");
        init(); // init handles the subscribes
    }
        
}
```

hirotakaster, 2017-11-06:
@natcl 
Thank you for your report. I think your code is good. 
client and server tcp connection is disconnected by several TCP's protocol causes, so your reconnect implementation will good for.

PR #91 (2021-05-14): Change keepalive instructions in readme
Hey,

After having some problems with disconnects I realised that the instructions I followed in the readme to increase keepalive time were not perfectly correct so I added some extra detail.

Fixes #69
closed 2021-05-14 by hirotakaster