hirotakaster/MQTT #70: MQTT_KEEPALIVE 15 is expensive

jrburk, 2017-12-27:
Can the constant

#define MQTT_KEEPALIVE 15

please been made configurable.  On mobile connects this costs a significant amount of money every month.

hirotakaster, 2017-12-28:
hi @jrburk 
you can change the KEEPALIVE on constructor.
`MQTT(char* domain, uint16_t port, int keepalive, void (*callback)(char*,uint8_t*,unsigned int));`

here is sample
  `MQTT client("server_name", 1883, 15 * 2, callback);`

jrburk, 2017-12-29:
Hello @hirotakaster 
a matching constructor cannot be found because I'm using it together with an extended message size:

MQTT cm(IOT_HOST, 1883, 5*60, callback, 1023);

I want to switch to your library MQTT-TLS at some point. Would it work there?

hirotakaster, 2018-01-04:
@jrburk 
okay, MQTT-TLS will work. I tested it with MQTT-TLS server.

closed 2018-03-20 by hirotakaster