hirotakaster/MQTT #73: Subscribe Retain Flag

marc-gist, 2018-08-03:
How can I check in the subscribe callback if the payload is a retained value?

Thanks,

hirotakaster, 2018-08-06:
@marc-gist 
check the MQTT protocol and your MQTT broker server specifications.
http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html

marc-gist, 2018-08-06:
No, I mean in your code and on the Photon/Core.

i.e. my code:

```
void callback(char* topic, byte* payload, unsigned int length) {
    char p[length + 1];
    memcpy(p, payload, length);
    p[length] = NULL;
    
    Serial.printlnf("Payload %s", p);
    if(!strcmp(p, "check")) {
        check_sensor();
    }
```
I can check the **topic** and **payload**, but how can I check the retained flag?

hirotakaster, 2018-08-07:
Now my callback don't have a retain flag argument as you know.
If you want to know the published message retain flag when your app first connect to the MQTT server, you could implements new callback method with checking the client.buffer mqtt header in loop() function.