hirotakaster/MQTT #78: client.connect() blocks argon in SYSTEM_THREAD(ENABLED)

Lurow, 2019-02-14:
Hi there,

I recently played around with the new "mesh" line from particle and used this lib.
At first, it was fine, everything was working as expected, but when we tried to enable threading and moved the `client.connect()` into the `loop(){}` block it appears to block all incoming connections.

```
if(!client.isConnected()){
            client.connect("bar");
            client.subscribe("foo");
}
```

is there any practice or solution for this problem?

best regards

Lurow, 2019-03-11:
Saw some problems with string IPs in one of the pull requests. Changing it to IPAddress resolved the issue

closed 2019-03-11 by Lurow

hirotakaster, 2019-03-12:
that's problem is already resolved in MQTT.h files.