hirotakaster/MQTT #80: Connecting and specifying LWT

bwieg92, 2019-06-09:
Can you provide an example of connecting and specifying a last will message?  I have tried using the function
`bool MQTT::connect(const char *id, const char *user, const char *pass, const char* willTopic, EMQTT_QOS willQos, uint8_t willRetain, const char* willMessage, bool cleanSession, MQTT_VERSION version)`
however I get errors about no matching function when I attempt to compile.  I have very carefully checked the data types I am passing to the function and believe I am using the correct types though.

hirotakaster, 2019-06-27:
hi, here is sample,
` 
client.connect("client", "id", "password", "last/will/topic", MQTT::QOS1, 0, "will message", true);
`

bwieg92, 2019-06-28:
Thanks, that worked!  I was missing the MQTT scope for willQos.

closed 2019-06-28 by bwieg92