hirotakaster/MQTT #87: _client.connected bug stops loop

KennPunt, 2021-01-02:
I am currently using MQTT on an electron.

In the loop I use the "MQTT::isConnected()" function around the "MQTT::loop".
It takes around 1 minute before the loop just stops working.

Button input still shows logs, but MQTT connection and logs in the loop function stop.
I have SYSTEM_THREAD enabled, but this worked for the past 2 years.

I put some logs around these methods:
```
bool MQTT::isConnected() {
    Log.info("TEST 1");
    bool rc = (int)_client.connected();
    Log.info("TEST 2");
    if (!rc) _client.stop();
    Log.info("TEST 3");
    return rc;
}
```
and logs always show this pattern:
```
0000028185 [moby] INFO: Cellular connection OK
0000028185 [app] INFO: TEST 1
0000028186 [app] INFO: TEST 2
0000028186 [app] INFO: TEST 3
0000028186 [moby] INFO: connected to Bremo
0000028286 [app] INFO: TEST 1
0000028287 [app] INFO: TEST 2
0000028287 [app] INFO: TEST 3
0000028798 [bremo] INFO: checkqueue :
0000029798 [moby] INFO: Cellular connection OK
0000029798 [app] INFO: TEST 1
0000029800 [app] INFO: TEST 2
0000029800 [app] INFO: TEST 3
0000029800 [moby] INFO: connected to Bremo
0000029900 [app] INFO: TEST 1
0000029900 [app] INFO: TEST 2
0000029900 [app] INFO: TEST 3
0000030412 [bremo] INFO: checkqueue :
0000031412 [moby] INFO: Cellular connection OK
0000031412 [app] INFO: TEST 1
0000033113 [app] INFO: TEST 2
0000033113 [app] INFO: TEST 3
0000033113 [moby] INFO: connected to Bremo
0000033213 [app] INFO: TEST 1
```
Where it always ends on TEST1...
I cannot find a way to enable try-catch functionality, so I don't know how I can debug this any further..

Please help

KennPunt, 2021-01-02:
Note: I included "HttpClient.h" too, to make a REST call before connecting to the MQTT server.
Can this lead to the TCPclient being in a buggy state?

KennPunt, 2021-01-02:
It appears to happen more frequently on firmware 2.0.X.
On 1.5.2 it appears to happen randomly around 10+minutes (and sometimes it even doesn't)
On 2.0.X it only can survive around 3 minutes.

hirotakaster, 2021-01-03:
@KennPunt I did not test on SYSTEM_THREAD enabled. I think more modification is required to this library on SYSTEM_THREAD.

hirotakaster, 2021-01-27:
Hi, now I upload SYSTEM_THREAD(ENABLED); version 0.4.30.
pls check threadtest sample app.

closed 2021-02-22 by hirotakaster