hirotakaster/MQTT #89: Message truncated at 255 bytes even though max packet size is 512 in constructor

platcham, 2021-02-20:
Hello,

I cannot seem to get the message size increase, even when I specify a larger maximum size in the constructor. 

Here is what I have at the beginning of my code: 
```c++
MQTT client(server, 1883, MQTT_DEFAULT_KEEPALIVE, callback, 512);   // connect to server on port 1883, default keepalive time, use callback function, allow 512 byte messages
```

And here is the function I am using to send an MQTT config:
```c++
void mqttSensorConfig(String name, String device_class, String state_topic, String unique_id, String unit_of_measurement, String config_topic) {
    //check if connection to MQTT server is still active
    if(!client.isConnected()) {                                         //if connection was not successful
        client.connect("weatherStationPhoton", MQTTUSER, MQTTPASS);     //reconnect to MQTT server, call ourselves "weatherStationPhoton"
    }
    
    char buf[400];                                                                  // create character buffer for data message
    JSONBufferWriter writer(buf, sizeof(buf)-1);                                    // create JSONBufferWriter object to write into the buffer
    writer.beginObject();                                                           // begin object creation  
        writer.name("name").value(name);                                                // write the name of the sensor
        writer.name("device_class").value(device_class);                                // write the type of sensor
        writer.name("state_topic").value(state_topic);                                  // write the mqtt topic where the state will be reported
        writer.name("unique_id").value(unique_id);                                      // write the unique ID for this sensor (MUST BE UNIQUE)
        writer.name("unit_of_measurement").value(unit_of_measurement);                  // write the unit of measurement for this sensor
        writer.name("device").beginObject();                                             // begin array of device characteristics
            writer.name("identifiers").value("230031000347343138333038");                   // identifier is Serial number
            writer.name("name").value("weatherStationPhoton");                              // write name of the device
            writer.name("model").value("Photon");                                           // write the model of the device
            writer.name("manufacturer").value("Particle");                                  // write the manufacturer of the device
            writer.name("sw_version").value(System.version());                              // write the device's current software version
        writer.endObject();                                                              // end array of device characteristics
    writer.endObject();                                                             // end the JSON object
    writer.buffer()[std::min(writer.bufferSize(), writer.dataSize())] = 0;          // null terminate buffer string
    
    // publish data to config topic
    client.publish(config_topic, buf);
    
    // loop client to publish data
    client.loop();
}
```

However, the message my broker receives is this:
```JSON
{"name":"Humidity Sensor","device_class":"humidity","state_topic":"weatherStation/humidity","unique_id":"weatherStation_humidity_sensor","unit_of_measurement":"%","device":{"identifiers":"230031000
```

When it should be this:
```JSON
{"name":"Humidity Sensor","device_class":"humidity","state_topic":"weatherStation/humidity","unique_id":"weatherStation_humidity_sensor","unit_of_measurement":"%","device":{"identifiers":"230031000347343138333038","name":"weatherStationPhoton","model":"Photon","manufacturer":"Particle","sw_version":"2.0.1"}}
```

As you can see, it is cutting the message off short, even though the 512 buffer should be plenty enough to transmit that message.

I've also attached the full version of my code, but other messages that are shorter send just fine, so I'm not sure what is happening.
[weatherStationMQTT.txt](https://github.com/hirotakaster/MQTT/files/6013918/weatherStationMQTT.txt)

Thank you for your time.

hirotakaster, 2021-02-22:
@platcham okay, I will check.

hirotakaster, 2021-02-22:
@platcham 
sorry, my readme is wrong. 
please update to this.

MQTT client(server, 1883, MQTT_DEFAULT_KEEPALIVE, callback, 512);
to 
MQTT client(server, 1883, 512, callback);

platcham, 2021-02-22:
That worked, thank you for the help!

closed 2021-02-22 by platcham