hirotakaster/MQTT #94: Connection timeout for attempting to connect to MQTT server?

shinglaifs, 2022-02-26:
Is there currently a way to set the connection timeout for attempting to connect to the MQTT server?
I realize that the current coding the mqttclient.connect() will take about 20 seconds before giving up on the attempting to connect to an IP that simply cannot be reach in the network.

hirotakaster, 2022-02-26:
Hi @shinglaifs ,
This library base network IO is [TCPClient](https://docs.particle.io/cards/firmware/tcpclient/connect/) , this TCPClient class don't have a network timeout option. And tcp timeout option is depend on particle firmware board&stack, If you want to set a timeout option you'll need to build custom firmware.

shinglaifs, 2022-02-26:
@hirotakaster thank you very much for your prompt response.
Right, I probably don't have the skill set to do a custom firmware.

However, I try to see if there is any other method that will work.
I come across the AsyncTCPclient example:
https://github.com/rickkas7/asynctcpclient/blob/master/AsyncTCPClient.cpp
from their threading tutortial: https://community.particle.io/t/particle-threads-tutorial/41362
Do you think this method (using threading and mutex) will work for the mqttclient.connect() ?

hirotakaster, 2022-02-26:
@shinglaifs Yes, AsyncTCPClient maybe work well, but I don't test with it and it difficult to check the problem as his comment 'Most APIs are not thread safe'. Because of that I think a lot of modification will be needed with AsyncTCPClient.
But I think 'Why server side IP endpoint not work?' is based MQTT server side HA(High Avaliabilty), so I think [emqtt](https://github.com/emqx/emqtt) or [mosquitto-cluster](https://github.com/hui6075/mosquitto-cluster) is good HA MQTT server for don't stop the endpoint IP address.

shinglaifs, 2022-02-26:
thank you @hirotakaster , I will give AsyncTCPClient method a try with your library.
And if it works, i will share it back here.